ArmoredTurtle/AT-Documentation #80: Wrong command in toolhead variable calculation

pascaldornfeld, 2025-07-27:
In here https://github.com/ArmoredTurtle/AT-Documentation/blob/main/docs/afc-klipper-add-on/toolhead/calculation.md#tool_stn-calculation it says:

> It may be easier to temporarily disable the AFC_POOP macro while you are adjusting the tool_stn value, as this will provide an easier way to observe the filament extruding from the hotend. Run AFC_TOGGLE_MACRO TOOL_CUT=0 to disable the macro, and AFC_TOGGLE_MACRO TOOL_CUT=1 to re-enable it after you have finished adjusting the value.

Probably the commands should be `AFC_TOGGLE_MACRO POOP=0` as the poop macro is mentioned in the first sentence.

pascaldornfeld, 2025-07-27:
Same later on the same page

> Don't forget to re-enable the AFC_POOP macro by running AFC_TOGGLE_MACRO TOOL_CUT=1 after you have finished adjusting the tool_stn value if you disabled it earlier.

ejsears, 2025-07-27:
Thanks, fixed.

ejsears, 2025-07-27:
Resolved in #81

closed 2025-07-27 by ejsears