ArmoredTurtle/AT-Documentation #85: [Feedback] AFC.cfg - /afc-klipper-add-on/configuration/AFC.cfg.html

adamcstorm, 2025-08-04:
Under Poop Settings, you're able to replace the default macro with a custom macro. Through testing, I've found that the only parameter that's passed to this macro is FLUSH_LENGTH. This means that any custom macro must accept this same parameter name if it does variable length purges. This is a bit of a hidden feature as it's not described anywhere. 

I suggest the comment under the Poop Settings section be improved to detail that a FLUSH_LENGTH parameter is passed to the custom macro. Optionally, it would be nice if it was clear that no other parameters are passed (if that's actually true - this is an assumption on my behalf).

adamcstorm, 2025-08-29:
@ejsears I believe that's likely where the call is being made. In gcode, I can pass any number of parameters down to another macro by using the {rawparams} argument. I had incorrectly assumed that's what was being done behind the scenes here. It is not clear to me that the only acceptable parameter name is purge_length (I mistakenly called this flush_length in the original comment, conflating the name of the parameter that's used in Orca). My suggestion in this case is to just make it clear that this is a valid option, but also the only valid option when using a custom poop macro.

ejsears, 2025-08-29:
@adamcstorm 

[image]

Think this is sufficient?

PR #97 (2025-08-29): Clarify AFC_POOP macro parameters and add 3D Lab Tech as a vendor
This pull request updates documentation for the AFC Klipper add-on configuration and BoxTurtle vendors list, focusing on clarifying macro usage and expanding vendor information.

Documentation improvements:

* Clarified that the only valid parameter for the `AFC_POOP` macro in `AFC.cfg.md` is `purge_length`, which specifies the length of filament to purge.

Vendor list update:

* Added `3D Lab Tech` as a Canadian vendor for the BoxTurtle kit in `vendors.md`.

Resolves #85