ArmoredTurtle/AT-Documentation #90: [Feedback] Calibration Process - /afc-klipper-add-on/installation/calibration.html

CharlesSampson, 2025-08-12:
There is an overall issue with the documentation where many pages assume the user has installed the Mainsail-AFC fork to enable the GUI. However, there is currently no documentation on how to install that, and the precompiled version is apparently only available to users with serials. In order to get a serial, you need a functioning unit, but getting a functioning unit without the GUI to guide you through calibration is currently confusing.

If the expectation is that users complete calibration just using default mainsail and manually activating the various calibration macros via console, then those steps really need to be described in the documentation.

ejsears, 2025-08-22:
@CharlesSampson I'm a bit confused, none of the calibration macros require any sort of integration. They can just be called through the console.

ejsears, 2025-09-05:
Closing due to no response.

closed 2025-09-05 by ejsears